opulo-inc/lumenpnp #225: [MOBO] - KiCAD won't load MOBO

Bazmundi, 2021-03-20:
**Motherboard Revision:** This should be on the front of your board in the format REV003 or REV003-???????

REV003 with @actions-user actions-user CI: Generate Gerbers and Associated Artifacts 7cf62c7 

**Issue Description:** Talk about the issue you're having here. 

Downloaded project.  Went to suck into KiCAD by double clicking on mobo.pro but KiCAD reported errors on start.
- What isn't working?
8:45:03 AM: file 'D:\Downloads\pnp\index-mobo-rev03\index-mobo-rev03\pnp\pcb\mobo\mobo.pro', line 1: '=' expected.
8:45:03 AM: file 'D:\Downloads\pnp\index-mobo-rev03\index-mobo-rev03\pnp\pcb\mobo\mobo.pro', line 2: '=' expected.
8:45:03 AM: file 'D:\Downloads\pnp\index-mobo-rev03\index-mobo-rev03\pnp\pcb\mobo\mobo.pro', line 3: '=' expected.

- What does it prevent or degrade?

Access to Mobo in KiCAD

- What have you done to diagnose that it is a hardware issue?

Nothing, its the Mobo pro files.

- Attach supporting files like software logs / scope images / logic captures / annotated scematic or pcb drawings.

**Suggested Fix:** Talk about how you think this should be fixed.
- What does the fix entail?
- Have you been able to test this fix?

daveismith, 2021-03-20:
Have you installed git lfs correctly?

daveismith, 2021-03-20:
To clarify, this typically happens if git lfs hasn't been installed an initialized as the files in the base git repo just contain a hash which references the file in the LFS storage. There are instructions in the wiki at https://github.com/sphawes/index/wiki/Contributing.

Bazmundi, 2021-03-21:
Ah! Something new to learn about git.  Thanks.

daveismith, 2021-03-21:
If it's something else let us know, otherwise please either close the ticket or let us know it can be closed.

Bazmundi, 2021-03-21:
There are instructions in the wiki at https://github.com/sphawes/index/wiki/Contributing.

closed 2021-03-21 by Bazmundi