opulo-inc/lumenpnp #227: [BUG] "Edit this page" feature in docs links to 404 page

darigovresearch, 2021-03-21:
The `Edit this page` button in the docs links to a URL/file which does not exist

Feel free to test this directly from the URL to replicate the issue - https://docs.index-machines.com/docs/

Would do a pull request but this docs system is new to us

daveismith, 2021-03-21:
This should now be resolved.

closed 2021-03-21 by daveismith

G-Pereira, 2021-03-22:
@darigovresearch When you can pease confirm this since you are an outside contributor.

darigovresearch, 2021-03-27:
@G-Pereira & @daveismith Yes now I think it redirects to a valid page, thanks for the help!

https://github.com/sphawes/index/edit/assembly-guide/doc/content/en/docs/_index.md

[image]