opulo-inc/lumenpnp #229: [CAD] - Tensioning system for X/Y/Z Belts

cad435, 2021-03-23:
In Order to get better precision, a belt tensioning system should be added. For the Y-Axis this should be added onto the front Index leg (opposite side of stepper motor)

however, as some people already printed the complete index frame (me included) I modeled a quick addon. This tensioner can be used instead of any existing belt clamp. (Intended for use on Y-Axis, NOT X-Axis!)
[image]

the left arrows point to two slots where two M3 nuts are inserted. Between the right blocks is a M3 screw. which has some 7x3x3 bearings onto it. A stack of M3 spacers also can be used instead of bearings.

.step an STL's attached, use it at your own risk!
[Tensioner_Fix.zip](https://github.com/sphawes/index/files/6192890/Tensioner_Fix.zip)

daveismith, 2021-04-04:
@cad435, what range of motion (in mm) does the tensioner need?

cad435, 2021-04-04:
About 2-3mm will do... More will be better, but I don't think necessary

darigovresearch, 2021-04-04:
Think this would be a great feature to add, over time belts lose their tension and having a way to do it in index would greatly reduce frustration after prolonged use!

darigovresearch, 2021-05-08:
Thanks @mhorsche thanks for taking a look at this, we haven't built an index yet so can't test it but as an initial release may be worth putting in a pull request depending on what the maintainers think. Do you have any photos of the part in use?

What do the maintainers/hardware builders think?

daveismith, 2021-05-08:
Stephen is also doing some tensioning work so hopefully between the two we
get something that works well.

MisterHW, 2021-05-16:
Exciting work! I'll leave my tensioning approach here for entertainment and inspiration. It certainly uses a minimal amount of fasteners, but it's only practical with custom sheet metal work in a certain volume.

[image]

https://hackaday.io/page/7198-hose-clamp-style-gt2-belt-tensioners

(timing belt crimp clamps are available, so please ignore the heat shrink tube)

maghostmil, 2021-05-24:
> In Order to get better precision, a belt tensioning system should be added. For the Y-Axis this should be added onto the front Index leg (opposite side of stepper motor)
> 
> however, as some people already printed the complete index frame (me included) I modeled a quick addon. This tensioner can be used instead of any existing belt clamp. (Intended for use on Y-Axis, NOT X-Axis!)
> [image]
> 
> the left arrows point to two slots where two M3 nuts are inserted. Between the right blocks is a M3 screw. which has some 7x3x3 bearings onto it. A stack of M3 spacers also can be used instead of bearings.
> 
> .step an STL's attached, use it at your own risk!
> [Tensioner_Fix.zip](https://github.com/sphawes/index/files/6192890/Tensioner_Fix.zip)
> 
> Edit: tension the belt, then
> move the axis and only after moving check the tensioning strength of the belt!

Can you add this .zip file again? It looks like expired - 404 Not found.

cad435, 2021-05-24:
@maghostmil 

[Tensioner.zip](https://github.com/index-machines/index/files/6532076/Tensioner.zip)

I'll close this issue now, as this is resolved by @sphawes in https://www.youtube.com/watch?v=EN0TxylB010&

closed 2021-05-24 by cad435

maghostmil, 2021-05-24:
> @maghostmil
> 
> [Tensioner.zip](https://github.com/index-machines/index/files/6532076/Tensioner.zip)
> 
> I'll close this issue now, as this is resolved by @sphawes in https://www.youtube.com/watch?v=EN0TxylB010&

Yes, but I completed assembly of earlier version and I need only this quick addon. Thanks.

cad435, 2021-05-25:
@maghostmil errm, I did a better job with this new tensioner addon...

[image]

[Y_Tensioner v5.zip](https://github.com/index-machines/index/files/6539810/Y_Tensioner.v5.zip)

sorry this came a day late... its better in nearly any aspect...

maghostmil, 2021-05-25:
Thanks @cad435. In the end, the solution from first post didn't work so well for me. When the bolt was tightened more, the addon broke at the nut. Today I will test your new version of this addon.

maghostmil, 2021-05-25:
@cad435 , I've tested this v5 addon, and works better than previous version, but I have one question for you: Wouldn't a 5mm hole be better (like hole in idler)? I feel micro-vibrations caused by idler isn't properly in whole hole.

cad435, 2021-05-25:
Well, I had only 3mm bore Idlers on hand, so I made it a M3 Bolt without any hesitation...

I may just forgotten, that the Index uses 5mm bore Idlers tbt.

But you can modify it (it's a .step, so any CAD will work) or drill out the hole...
I might do it tomorrow if I don't forget it till then :)

PR #318 (2021-06-13): Belt Tensioning Update and Mechanical BOM Refresh
Hey all!

Here's a monster PR with all the recent CAD updates, along with an updated mechanical Bill of Materials. Here is what is included:

- Fixed improper method of building files in FreeCAD, now only basing sketches off of Datum Planes
- All belts have fine tension adjustment, either through a screw adjustment or a slotted holes for a stepper motor mount (Closes #229)
- All roller gantries have fine tensioning on both sides of the bolt holding the third roller
- All fasteners are included in the assembly file, generated by the Fastener wor [... cut]