opulo-inc/lumenpnp #231: [MOBO] - Three RS-485 Ports

sphawes, 2021-03-25:
**Motherboard Revision:** REV03

**Enhancement Description:**
- Add in two extra RS-485 ports so that there are three total.
- These extra two are for communication with other machines (such as a hopper or reflow oven), or for allowing folks to hack their own modules and machines into the Index.
- The two extra ports should have 5V, GND, and the two RS-485 lines.
- The main RS-485 port for the feeder should use a 2x3 2.54mm IDC connector (same component and pinout as from Mobo REV02)

G-Pereira, 2021-03-25:
What about adding just two more connectors to the existing bus? We seem to be targeting 2 more devices so we are just subtracting 2 from the maximum number of devices on a single bus which is 256. People can still use this for their own things

daveismith, 2021-03-25:
My concern about using the same bus would be that we don't know the protocol and bandwidth requirements of these future expansions, and how compatible they would be with the protocol that's been defined for the feeders. Having separate busses allows for flexibility in that department.

For example, the current feeder protocol starts with an address byte and assumes a certain amount of quiet time between the frames to allow devices to synchronize. A oven or hopper protocol may use something completely different, especially if we wanted to integrate an off the shelf solution.

G-Pereira, 2021-03-25:
The RS-485 chip is not particularly cheap So I would say we should try to standardize our protocol for all the boards we use and keep a single bus. I don't think we'll want to use a COTS motherboard for the hopper or reflow oven to keep the entire machine on the opensource spectrum.

daveismith, 2021-03-25:
Could the connectors and circuitry for the additional two RS485 ports be part of a variant so people could select whether they need this additional functionality or not?

G-Pereira, 2021-03-25:
I think you already know my opinion on that :smile: I would say people can use the AUX port for additional RS-485 transceivers. From my perspective adding them on board is less 4 pins each that we can expose on the AUX port that people have full flexibility upon

adlerweb, 2021-03-25:
> Could the connectors and circuitry for the additional two RS485 ports be part of a variant

...or just expose the required pins to a header and allow for future expansion using an additional plug in board

G-Pereira, 2021-03-25:
I don't know you @adlerweb but you seem like a nice guy :smile:

daveismith, 2021-03-25:
@G-Pereira - I guess my concern is if you put everything into the AUX port it makes it difficult to experiment or expand if you're using what will eventually become part of the standard set of machines.

@adlerweb - I considered suggesting something similar to the form factor we're using for motor drivers, but then you've got the extra overhead of sourcing another board to consider. I'm not sure what people's thoughts on that would be.

G-Pereira, 2021-03-25:
@daveismith What I am saying is that what we plan that people might use is probably not the case because people usually come up with awesome things entirely different. That's why it is so important to give some thought about the pinout of the AUX connector, which Stephen has done an amazing job.

daveismith, 2021-03-25:
@G-Pereira I'm not sure we're disagreeing. What I'm trying to say is that I think it is more flexible and beneficial for users to keep the 485 expansion ports separate from the aux port. This means that if I have an expansion using only 485 it doesn't impede access to an aux port.

G-Pereira, 2021-03-26:
Ah! Then we're not talking about the same thing! What I am suggesting is instead of hardwiring pins **x**,**y**,**w** and **z** (4 pins for controlling the transceiver) resulting in the **a** and **b** bus pins we can expose **x**,**y**,**w** and **z** (not a and b) and don't include the transceiver on board.
However, so we can still use the bus for other things (reflow oven and hopper) we should add more connectors to the transceiver we have on board.

cad435, 2021-03-30:
Personally I'd suggest a middle route... using one RS-485 transcreiver exclusively for the feeder and using another independent transcriever with 2 attached plugs on the mobo for everything else.

Also, as @sphawes opened this request he should have the last decision, as I think it has to do with what he wants to do next with the index-'ecosystem' (after the conveyor) which we might not be aware of at this point...

daveismith, 2021-03-30:
@cad435 - We're definitely keeping that in mind and discussing with @sphawes to ensure that the requirements for the future vision are met. 

I think discussion around the purpose of a feature or the best way to implement it actually makes for a better project as we all sometimes have a solution in mind which when iterated on with others can be improved. In discussion on discord for example, the SMEMA specification was discussed which is an industry standard way of handling the objective that Stephen is trying to achieve so it makes sense to ensure we review that with him to see if we can implement his vision in a way that's more cost effective.

G-Pereira, 2021-03-30:
According to yesterday's discussion, we are keeping only one bus with only one connector