opulo-inc/lumenpnp #235: [MOBO] - Buck Converter Part Choice

sphawes, 2021-03-25:
**Motherboard Revision:** REV03

**Context:** 
- The REV02 Mobo used a TPS54336, rated for 3 amp output. This is quite overkill if we aren't powering the feeders through it anymore. 
- The current buck converter IC in REV03 Beta is TPS54531. This is rated for a bit more current output, and is a bit harder to source.

**Enhancement Description:** 
- Pick and implement a replacement buck converter that 1) Is easily sourced through Digikey/LCSC/Mouser, 2) reasonable price (up to a couple bucks), 3) has a low number of required components to implement, and 4) has a current output rating of at least 2 amps (accounts for 500ma USB spec for the three USB ports, then 500ma for the LEDs, and powering the 3.3v LDO).

darigovresearch, 2021-04-02:
Should this issue be closed if pull request #254 was merged?

Not sure if automation is setup for this specific project board or if it's remaining open for another reason

G-Pereira, 2021-04-02:
Thank you @darigovresearch !

closed 2021-04-02 by G-Pereira

darigovresearch, 2021-04-02:
@G-Pereira you're very welcome!