opulo-inc/lumenpnp #237: Feeder TT Motor Mount Does Not Fit

sphawes, 2021-03-26:
In Feeder REV04 and REV05, the TT motor does not seat properly in the motor mount print. There are a few tabs on the motor that interfere, where a relief should be added to the model.

This is also just for this one specific TT motor, of which there are many small variants. Also considering what dimensions of a TT style motor are standard across brands would be good to use as the mating features of the motor mount, to allow compatibility with the most number of motors.

daveismith, 2021-03-30:
I would agree and note that even with the tab removed the current design causes issues because the motor isn't completely supported so clamping from the top can actually make the motor body come out of parallel with the PCB quite easily. I would suggest a cutout for the tab and a step up to match the difference between case size of the motor and gear box.

G-Pereira, 2021-05-04:
Does this require changes bon the PCB?

daveismith, 2021-05-04:
@G-Pereira, no. The change is in the plastics

G-Pereira, 2022-03-05:
The feeder-topology branch adresses this but it is still not ready to merge because there's topology errors in many of the files

sphawes, 2022-05-11:
Revamping all feeder files where this will be addressed, closing.

closed 2022-05-11 by sphawes