opulo-inc/lumenpnp #240: [BUG] References to RING in feeder page

darigovresearch, 2021-03-27:
It appears that the readme for the feeder contains references to RING. After watching the [development playlist](https://www.youtube.com/playlist?list=PLIeJXmcg1baLBz3x0nCDqkYpKs2IWGHk4) our understanding was that RING was deprecated over using RS-485.

Should this be updated or were we mistaken?

Happy to make a pull request to resolve if this should be switched out!

Page in question - https://github.com/sphawes/index/tree/master/feeder

cad435, 2021-03-30:
yes, that is indeed an error...

@daveismith might be the person to change this or approve a pull request regarding this :)

darigovresearch, 2021-04-02:
Thanks @cad435 for the guidance, will put in a pull request now!

darigovresearch, 2021-04-02:
Merging pull request #257 would resolve this issue

closed 2021-04-02 by daveismith