opulo-inc/lumenpnp #241: [BUG] Dead link to BOM in Docs

darigovresearch, 2021-03-27:
There appears to be a link to a CSV version of the BOM on the assembly page docs.

Am assuming that the new correct URL is https://github.com/sphawes/index/blob/master/pnp/bom.csv

Happy to make the update if given the confirmation that the above is correct!

Docs page in question - https://docs.index-machines.com/docs/mechanical-assembly/
Dead link URL - https://github.com/sphawes/index/blob/freecad/pnp/bom.csv

daveismith, 2021-03-27:
That’s the correct file to link to but please note it’s currently out of
date and you would probably get a more accurate idea from the sections at
the top of the assembly guide.

darigovresearch, 2021-03-28:
@daveismith thanks for the guidance, pull request #244 would resolve the issue & we have highlighted your note as well.

Feel free to adjust if required!

closed 2021-03-28 by daveismith