opulo-inc/lumenpnp #243: [DOCS] License for the documentation

darigovresearch, 2021-03-28:
Various open source projects have different licenses for the product itself and the documentation. Often it's a CC-BY or other creative commons license for the documentation where the code and hardware is under a different suitable software license. We couldn't find any for the [docs themselves](https://docs.index-machines.com/docs/). Depending on how you've built them you can autogenerate a link to the license at the bottom of each page.

Have you thought on this topic and if so which licenses were you considering?

Having an explicit license may mean you will get more open source contributions to the docs by the open source hardware and open source software community.

daveismith, 2021-03-30:
Currently my understanding is everything is licensed under GPL3, but it's probably something that should be reviewed to make sure that it's correct. I know for example that typically hardware uses different licenses such as CERN OHL (see this blog post [blog post](https://opensource.com/law/15/2/intro-open-hardware-licensing)). You also raise a good question about doc licensing. 

I've assigned this to @sphawes to review and make a decision. Once he's either confirmed GPLv3 or something else I can update the docs to add that info at the bottom of the generated pages.

darigovresearch, 2021-04-02:
@daveismith great point, some more mature open source projects have three licenses which may apply here
- One for the software (i.e. GPL, MIT etc)
- One for the hardware (i.e. CC0, CERN, TAPR etc)
- And one specifically for the documentation (CC-BY etc)

Examples of good similar open source projects which do this well are 
- [MaslowCNC](https://www.maslowcnc.com/) an open source vertical CNC machine for cutting large items
- [Farmbot](https://farm.bot/pages/open-source) an open source machine for seeding, watering and monitoring farming machine

The above have the same kind of components you have, PCB designs, STL/CAD models and software so you may find some benefit from seeing what they did. 

I think this issue is quite an important issue as from experience depending on the license chosen correlates to the number of community contributions and the longevity of the project as a whole

Happy to discuss if we can be of any assistance

darigovresearch, 2021-05-01:
For the record, happy to put our contributions under whatever license you choose in the end!

The earlier this is decided & made explicit, the more likely you will prevent issues like https://github.com/openpnp/openpnp/issues/1163 of having contributions which aren't under a unified license

sphawes, 2022-05-11:
We'll be moving the docs to a separate repo than the source shortly, and we'll make sure to have a dedicated license for them! Thanks for bringing this up @darigovresearch! I'll close this once that happens.

darigovresearch, 2022-05-14:
You're very welcome & sounds good!

closed 2023-08-30 by G-Pereira