opulo-inc/lumenpnp #247: [DOCS] Order of sections

darigovresearch, 2021-03-28:
Is it possible to adjust the location of the [Reference](https://docs.index-machines.com/docs/reference/) section in the left hand panel of the docs so that it comes last like an appendix?

Seems like having the following order would work well for assembly order required for Index
- Mechanical Assembly
- Electrical Assembly
- OpenPNP Setup
- Reference

Happy to make a pull request (given you agree with the suggestion above) if shown what to change and where but currently not sure how the docs are generated

daveismith, 2021-03-28:
The order is changed with the weight in the header section of the relevant _index.md files. 

Generation is done with Hugo doc generator and the docsy theme. Take a look at https://www.docsy.dev if you want to get set up for local preview. I haven't gotten to writing a guide of how to contribute, but the actual updates to the site happen automatically on merge to the assembly guide branch via GitHub actions. So contributing pull requests is the proper way to update the site (it just happens).

darigovresearch, 2021-03-28:
Pull request #248 would resolve this issue if merged

closed 2021-03-28 by daveismith

darigovresearch, 2021-03-28:
Issue #251 creates an initial documentation page in the reference section outlining how the documentation works