opulo-inc/lumenpnp #258: [DOCS] No links or explanation of print setting in docs

darigovresearch, 2021-04-02:
Current explanation on print settings can be found in the readme here - https://github.com/sphawes/index#3d-printing

Thought it may be useful to have this in the docs as a page in the reference section which potentially could be made more complete than in the readme later.

Also an initial link can be placed where it's first mentioned in the mechanical assembly section [here](https://docs.index-machines.com/docs/mechanical-assembly/) but it may get lost if not visible with a separate subsection in the reference.

For context here is a good example of printing guidelines set up as a separate page which later makes assembly much easier for [openflexure](https://build.openflexure.org/openflexure-microscope/v6.1.5/docs/#/0_printing) an open source Raspberry Pi based microscope. You may not need something so elaborate for now.

Happy to make the relevant pull request if given guidance on what the team think would be best!

daveismith, 2021-04-02:
I think a separate page for printing would make a lot of sense since we're already realizing we probably need to give guidance on recommended print orientations to ensure the proper strength for each part.

darigovresearch, 2021-04-02:
Great an initial page has been setup and would resolve this issue if pull request #259 is merged.

@daveismith great point on the orientation for printing, one trick that we've seen is that when you export the model to .stl, you orient it as it would on the build plate so the user doesn't need to mess with orientation.

I also think that if you design parts to not need supports that would be great as that can be difficult for some users. Saying on the print settings page that the prints do not need supports encourages people that the 3D printed parts are easier to print as there are no supports which result in "wasted" material and frustration in removal.

If supports are required; designing the supports in the STL also provides a convenience factor and removes the unknown of if auto-generated slicer supports will work or not.

I am also unsure how to add a cross-documentation link on the [Machine assembly page](https://docs.index-machines.com/docs/mechanical-assembly/) so feel free to append it to the pull request if you know how!

daveismith, 2021-04-04:
I made an adjustment to your branch but got it wrong, so I've committed the corrected version directly. You can see more at [https://gohugo.io/content-management/cross-references/](https://gohugo.io/content-management/cross-references/).

closed 2021-04-04 by daveismith