opulo-inc/lumenpnp #261: [FEATURE REQUEST] Make glossary definitions linkable in docs

darigovresearch, 2021-04-03:
Very often in glossaries for open source projects you can link to a specific definition to the glossary. Thought this would be a nice feature but unsure of how to implement this with the documentation framework being used.

This way if there is some jargon in another section of the documentation it can directly be linked to the definition in the glossary to make it easier for new joiners to understand.

daveismith, 2021-04-04:
Check out the relative linking info: https://gohugo.io/content-management/cross-references/. I'm not quite sure how to put a custom anchor into the glossary.

darigovresearch, 2021-04-05:
@daveismith Thanks for the guidance we'll take a look but perhaps if you put the labels `good first issue` and `help wanted` someone else may be more likely to come across it and solve the issue in a pull request before we do

darigovresearch, 2021-04-05:
Pull request #268 attempts to add anchors to the glossary but haven't managed to figure out how to build the docs to test so feel free to test whenever you have a spare minute if they can be referenced from another page

sphawes, 2021-09-20:
@daveismith Can I close this? Did merging #268 add this in?

closed 2021-09-20 by daveismith