opulo-inc/lumenpnp #265: [CAD] - Spool Motor Wiring Improvement

daveismith, 2021-04-05:
The spool motor wiring goes through a hole in the guide-right portion of the feeder. Unfortunately this means that you cannot then take the feeder apart after assembly without de-soldering the motor wires.

[image]

I would suggest that some sort of exit be modelled into the CAD to allow for the plastic part to be removed / replaced without de-soldering the motor.

jonbobcar, 2021-05-25:
Do the motor wires go to those two header pins below the motor? If yes, then why cant the motor be connected to those with a female header which could be pulled out and the motor removed from the assembly?

daveismith, 2021-05-25:
It might be able to, but the space is fairly tight and as such I worry that adding a header and pins we could actually have problems with people being able to reliably assemble this. 

Moving the header location may be possible but would require a re-layout of most of the traces connecting to that portion of the board.

jonbobcar, 2021-05-26:
@daveismith, ok great info. I agree that modifying the mechanical enclosure is a better option here.

sphawes, 2022-05-11:
Feeder is getting revamped for REV06, this is out of date, closing.

closed 2022-05-11 by sphawes