opulo-inc/lumenpnp #275: Mechanical Left Side - process improvement

NT7S, 2021-04-11:
_Also applies to Right Side_

In my experience in assembling the frame, I think that step 2 (inserting all of the slot nuts at the same time) is too unwieldy to be practical. I tried doing it that way and had various slot nuts falling out as I was trying to secure others.

I believe it would better to place the 2020 extrusion into the FDM frame pieces, and then put slot nuts in as needed for each following steps 4-6, securing the nuts with the screws before moving onto the next. The final two nuts needed for slot E could be added as a final step.

daveismith, 2021-04-11:
Hi Jason,
I think it depends on your slot nuts. There are a type that you can just
insert from the slot top and tighten to fasten. There are other styles that
must be inserted from the end. Do you know which style you were using? We
might need to add some notes.

NT7S, 2021-04-11:
Ah, I'm still a bit of a novice when it comes to building things using aluminum extrusion, so I didn't know there were captive slot nuts. I have the ones that can slip in from the top of the slot, which is why they were falling out when I was trying to do it the way depicted in the docs. I agree that it might be good to either spec the captive nuts if you can or make notes about alternate build instructions if you have the slot nuts like mine.

paulsykes, 2021-04-13:
Ahh this makes more sense now. The nuts I bought were also not captive. Does captive or non-captive have any advantage. It might be worthwhile standardising on one type for the MVP documentation, just to keep things simple? I bought the nuts linked in the bom.csv

daveismith, 2021-04-15:
@NT7S  / @paulsykes  - could you take a look to see if the updates I've made make more sense. There's a note on the mechanical assembly page as well some notes in the left and right assembly pages.

closed 2021-04-15 by daveismith

NT7S, 2021-04-15:
Thanks, looks good!

paulsykes, 2021-04-20:
Looks good to me, many thanks.

commit 2a828cbf (2021-04-15): Fixes #275, #278 as well as extending some of the instructions for the x-gantry assembly.