opulo-inc/lumenpnp #276: Feeder bus wires might have been swapped in a recent commit by mistake

G-Pereira, 2021-04-11:
This can either be at the mobo, feeder or feeder floor

daveismith, 2021-04-13:
Here's the current setup

Mobo Rev 03 U14 | Mobo Rev 03 J17 | Feeder Floor Rev 05 J1 | Feeder Floor Rev 05 J2 | Feeder Mobo Rev05 J3 | Feeder Mobo Rev 05 U7
------------------|-------------------|------------------------|------------------------|-------------------------|--------
A (RS-485+) | **5** (RS-485+) | **6** (RS-485+) | 4/7 (RS-485+) | 2 (RS-485 +) | A (RS-485 +)
B (RS-485-) | **6** (RS-485-) | **5** (RS-485+) | 3/8 (RS-485-) | 3 (RS-485 -) | B (RS-485 -)

Mapping from Feeder Floor to Feeder
Signal | Floor | Feeder
-------|------|-------
GND | 1/10 | 5
1WIRE | 2/9 | 4
RS-485- | 3/8 | 3
RS-485+ | 4/7 | 2
VIN | 5/6 | 1

So, looking at all of this, it seems like there's an issue in the connection between the Mobo J17 and feeder floor J1. ie, pins 5&6 as swapped. 

@G-Pereira, I would suggest you swap pins 5 & 6 on the feeder floor J1

G-Pereira, 2021-04-13:
You have two pluses at the feeder floor column, can you fix that?

daveismith, 2021-04-13:
Done. See above comment.

G-Pereira, 2021-04-14:
Swapped signals on feeder floor fixed!

closed 2021-04-14 by G-Pereira