opulo-inc/lumenpnp #278: [DOCS] Y Gantry - Error/clarification

NT7S, 2021-04-13:
Step 16 references screws inserted in Step 8, but I believe that reference is stale or incorrect. The type of screws here are not noted, but I believe they are M5x10.

Also, Step 23 is very unclear. An illustration of the end configuration would be incredibly helpful here.

daveismith, 2021-04-14:
Agreed that this could be clearer. For step 23 in particular, there’s two
separate parts that get mounted. One is the X motor mount which has the
stepper screwed on from the x motor
Mount stage. The other is an umbilical mount. You use the m5x25 screws to
go through the umbilical mount screw holes and then through the motor mount
and then finally through the gantry to nuts.

The umbilical mount should be pointing upwards. If you look at the photo on
http://docs.indexmachines.io/docs/ you can see it sticking up behind the
motor on the left side of the machine.

With all that said, I’ll try to better capture this soon in the docs.

daveismith, 2021-04-15:
@NT7S - could you check to see if it's a bit clearer with the latest update. If so, please close the issue or let me know it can be closed.

closed 2021-04-15 by daveismith

NT7S, 2021-04-15:
Much better, thank you