opulo-inc/lumenpnp #281: [DOCS] Docker issue

darigovresearch, 2021-04-17:
When trying to build the docs to resolve issue #261 from instructions given in pull request #271 we get the following error when running `docker-compose build`

Could anyone advise?

```Traceback (most recent call last):
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\site-packages\docker\api\client.py", line 214, in _retrieve_server_version
    return self.version(api_version=False)["ApiVersion"]
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\site-packages\docker\api\daemon.py", line 181, in version
    return self._result(self._get(url), json=True)
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\site-packages\docker\utils\decorators.py", line 46, in inner
    return f(self, *args, **kwargs)
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\site-packages\docker\api\client.py", line 237, in _get
    return self.get(url, **self._set_request_timeout(kwargs))
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\site-packages\requests\sessions.py", line 546, in get
    return self.request('GET', url, **kwargs)
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\site-packages\requests\sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\site-packages\requests\sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\site-packages\requests\adapters.py", line 449, in send
    timeout=timeout
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\site-packages\urllib3\connectionpool.py", line 603, in urlopen
    chunked=chunked)
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\site-packages\urllib3\connectionpool.py", line 355, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\http\client.py", line 1229, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\http\client.py", line 1275, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\http\client.py", line 1224, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\http\client.py", line 1016, in _send_output
    self.send(msg)
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\http\client.py", line 956, in send
    self.connect()
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\site-packages\docker\transport\npipeconn.py", line 32, in connect
    sock.connect(self.npipe_path)
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\site-packages\docker\transport\npipesocket.py", line 23, in wrapped
    return f(self, *args, **kwargs)
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\site-packages\docker\transport\npipesocket.py", line 72, in connect
    raise e
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\site-packages\docker\transport\npipesocket.py", line 59, in connect
    0
pywintypes.error: (2, 'CreateFile', 'The system cannot find the file specified.')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\darigov\AppData\Local\Programs\Python\Python37-32\Scripts\docker-compose.exe\__main__.py", line 7, in <module>
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\site-packages\compose\cli\main.py", line 81, in main
    command_func()
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\site-packages\compose\cli\main.py", line 200, in perform_command
    project = project_from_options('.', options)
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\site-packages\compose\cli\command.py", line 70, in project_from_options
    enabled_profiles=get_profiles_from_options(options, environment)
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\site-packages\compose\cli\command.py", line 153, in get_project
    verbose=verbose, version=api_version, context=context, environment=environment
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\site-packages\compose\cli\docker_client.py", line 43, in get_client
    environment=environment, tls_version=get_tls_version(environment)
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\site-packages\compose\cli\docker_client.py", line 170, in docker_client
    client = APIClient(use_ssh_client=not use_paramiko_ssh, **kwargs)
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\site-packages\docker\api\client.py", line 197, in __init__
    self._version = self._retrieve_server_version()
  File "c:\users\darigov\appdata\local\programs\python\python37-32\lib\site-packages\docker\api\client.py", line 222, in _retrieve_server_version
    'Error while fetching server API version: {0}'.format(e)
docker.errors.DockerException: Error while fetching server API version: (2, 'CreateFile', 'The system cannot find the file specified.')

daveismith, 2021-04-19:
@darigovresearch - have you used docker for other things? It's odd that it's throwing these errors since they aren't inside the docker container.

darigovresearch, 2021-04-24:
@daveismith first time user of docker, if you have any preference on installation or instruction on how you installed it may help others not have this issue or make it easier to get started

For context working on Windows and tried to install docker via pip as the UI install seemed to want to install more advanced features of docker that didn't seem necessary at the time

daveismith, 2021-04-25:
@darigovresearch, can you clarify how you installed with pip? I'm wondering if you ended up just installing a docker client with no actual docker daemon.

The docker approach is the simplest, but you could look into getting something from https://www.docsy.dev/docs/getting-started/ working (this is the theme we're using) and documenting that.

darigovresearch, 2021-04-25:
@daveismith went through the docker installation for docker compose directly from the docs and not the docsy documentation. See the `Alternative install options` on the following page https://docs.docker.com/compose/install/

Thought the docker engine was also installable via pip but evidently not, the docsy getting started certainly helps, potentially worth linking it and that should suffice

sphawes, 2021-10-05:
@darigovresearch Trying to close out old tickets, is this resolved? Was it a problem with your Docker install?

darigovresearch, 2021-10-16:
@daveismith & @sphawes most likely a docker installation issue, I'll close it but keep it in mind if you see someone else raising a similar issue in the future.

This was a first time attempting to use docker & thought it would be worth a shot

closed 2021-10-16 by darigovresearch