opulo-inc/lumenpnp #282: Idler pulleys - stopping the rotating surface rubbing when mounted horizontally.

spavinm, 2021-04-17:
Mounting the idler pulleys horizontal causes the rotating pully wall to run on the surface of the fixing. This proposed change applies to FDM_0001_00_front_left_leg, FDM_0002_00_front_right_leg and FDM_0013_00_idler_mount. 

As the inner race of the proposed/designed in idler pulleys is recessed into the part, a boss is required to lift the rotating surface clear of the lower surface of the fixing. The current left and right front legs have a 1mm clearance on the pully, the two proposed bosses (one top and one bottom) should be added to all three parts. The boss dimensions are 0.5mm high and the flat top surface 0.5mm wide. No increase in the clearance is required.

[FDM_0001_00_front_left_leg-plus-idler-bearers.pdf](https://github.com/index-machines/index/files/6329593/FDM_0001_00_front_left_leg-plus-idler-bearers.pdf)

ocouch, 2021-04-20:
Also see relevant conversation in Discord:
https://discord.com/channels/688723670461448225/739250421893300264/832992708955078696

spavinm, 2021-04-20:
Here is picture of a modified FDM-0013-00 and the idler pully is now clear of the surface and rotates freely.
[image]

stuartpittaway, 2021-07-02:
Is this bug still relevant given the latest design with tensioner ?

G-Pereira, 2021-07-03:
> Is this bug still relevant given the latest design with tensioner ?

It is, nothing keeps the idler from rubbing on the print

stuartpittaway, 2021-07-03:
> > Is this bug still relevant given the latest design with tensioner ?
> 
> It is, nothing keeps the idler from rubbing on the print

Ah, I used a metal washer to stop that, although I also found that the pully actually centres itself in the hole and doesn't actually touch the sides when the belt is under tension.

sphawes, 2021-09-20:
I've found the same as Stuart here. Going to close this unless someone experiences a specific problem with this.

closed 2021-09-20 by sphawes