opulo-inc/lumenpnp #286: FreeCAD and KiCAD preferred versions

SageRobots, 2021-04-23:
The CONTRIBUTING.md file should include versions so everyone interested in contributing can be sure to use the same.

For example,

"CAD is done in FreeCAD version 0.19.1, available on the FreeCAD releases page https://github.com/FreeCAD/FreeCAD/releases"

darigovresearch, 2021-04-24:
Can agree that this would be helpful. Doesn't need to be exact depending on if there are any breaking changes in the project structure for different versions of the CAD programs.

For example,

"PCB design was developed in KiCad 5.1.9 but should work for versions 5.1 onwards"

G-Pereira, 2021-04-25:
Thank you! If I recall correctly we are using symbols that were introduced on KiCAD 5.1.9 so I think it's best to claim only 5.1.9. On the other hand, for FreeCAD we only need the version whichever introduced the Hole feature or above

darigovresearch, 2021-05-08:
Thanks @G-Pereira, a [pull request for KiCad version](https://github.com/index-machines/index/pull/308) has been added.

What FreeCAD version are you currently using & what are the versions which other maintainers & contributors using?

That could be easier to find than finding the specific version of a feature in the FreeCAD changelogs

G-Pereira, 2021-09-14:
This was fixed by #308

closed 2021-09-14 by G-Pereira