opulo-inc/lumenpnp #288: Make feeder blink over OpenPnP

G-Pereira, 2021-04-25:
To help identify feeder configured on OpenPnP a command can be sent to a feeder to make it blink.

Jnesselr, 2021-08-12:
@daveismith How hard would it be to refactor the existing feeder code to make the feeder do an initialization lighting sequence. ie on "init feeder"?

daveismith, 2021-08-12:
@Jnesselr - I take it you mean when it receives an Initialize Feeder protocol command rather than when it first boots?

I'll have to abstract it out a bit but it should be possible. 

Do we want LED sequences for other commands and should we discuss the addition of an "Identify" command?

Jnesselr, 2021-08-15:
Identify might be a useful command to add, and I wouldn't mind it being part of the v1 protocol either at this point, since v1 isn't really released yet.

I was thinking through some options as far as when and how to light it. Right now, we only call initialize feeder in a "prepare for job" setting or if the feeder gets moved and we have to find it again. The effect I was thinking of was when I was searching, strobing the feeder lights so if you had a bank of feeders, it would just go through the feeders and light each one up one by one for a second or so. Granted, I would want the feeder to turn off the lights itself. It's more of a visual effect of "yes, that feeder got noticed by OpenPnP" so if you see one that didn't light up, you'll notice it faster than when you're trying to find it in the feeder list.

sphawes, 2022-05-11:
Going to punt on this for now. Initializing a feeder causing a light sequence would cause quite a lot of light sequences based on usage of the protocol so far. It could be useful to add this button into OpenPnP and the protocol, but until it becomes apparent that it's worth it, we'll close this.

closed 2022-05-11 by sphawes