opulo-inc/lumenpnp #292: [CAD] - Automatically export DXF files like STL files

Jnesselr, 2021-04-27:
We export STL files from the freecad model. It'd be awesome to export the DXF files for the build plate. It might also be useful to have multiple options for the build plate that are all exported.

stuartpittaway, 2022-02-01:
I've included in cad-staging a script to generate DXF files for any Freecad model file in the CSM folder.

This DXF output is also included in the generate-3d-prints.yaml GITHUB action to include the output in the artifact ZIP which is generated.

HOWEVER...

I can't get it to run from the GITHUB action.  Originally the Python libs were complaining about missing QT, so I installed that, but then its now complaining about undefined symbol.

Out of my depth on this one - any help @sphawes  or @Jnesselr  or @daveismith ?

```
Added possible FreeCAD path: /home/runner/work/index/index/squashfs-root/usr/lib
Traceback (most recent call last):
  File "/home/runner/work/index/index/pnp/cad/export-csm.py", line 29, in <module>
    import FreeCAD
ImportError: /home/runner/work/index/index/squashfs-root/usr/lib/./libFreeCADBase.so: undefined symbol: _PyTraceMalloc_NewReference
Error: Process completed with exit code 1.
```

stuartpittaway, 2022-02-01:
Made a few changes, different error but still not working (script runs perfectly on local machine).

```
Added possible FreeCAD path: /home/runner/work/index/index/squashfs-root/usr/lib
Traceback (most recent call last):
  File "export-csm.py", line 30, in <module>
    import importDXF
  File "/home/runner/work/index/index/squashfs-root/usr/Mod/Draft/importDXF.py", line 56, in <module>
    import Part, Draft, Mesh
  File "/home/runner/work/index/index/squashfs-root/usr/Mod/Draft/Draft.py", line 53, in <module>
    from draftutils.utils import ARROW_TYPES as arrowtypes
  File "/home/runner/work/index/index/squashfs-root/usr/Mod/Draft/draftutils/utils.py", line 39, in <module>
    import PySide.QtCore as QtCore
  File "/home/runner/work/index/index/squashfs-root/usr/Ext/PySide/QtCore.py", line 1, in <module>
    from PySide2.QtCore import *
  File "/home/runner/.local/lib/python3.8/site-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
ImportError: /home/runner/work/index/index/squashfs-root/usr/lib/././libQt5Core.so.5: version `Qt_5.15' not found (required by /home/runner/.local/lib/python3.8/site-packages/PySide2/QtCore.abi3.so)
```

sphawes, 2022-05-11:
This done and working! Closing

closed 2022-05-11 by sphawes