opulo-inc/lumenpnp #293: [MOBO] - U12 step down to 5V is connected to ground through a single via.

paulsykes, 2021-04-28:
REV003

Pin 7 of U12, the ground connection for the step down DC to DC for the 5V rail, is connected to ground through a single via. For reliability / performance / emissions improvements, I would recommend adding additional via / vias, or improving the route to gnd in some other way.

TPS54331EVM datasheet shows an ideal reference layout. Our current demands are relatively low, so we have some flexibility in regards layout.

daveismith, 2021-04-28:
@G-Pereira - Swapping the locations of R26 and R28 would allow you to shorted the trace running from pin 5 of U12 to pin 1 of R27 (since you could connect it to pin 2 of R26). It would then allow there to be a ground pour that exited both sides of the U12 pin 7 and connected via the GND pin of R28 to a second via. It would also give more space for another via to be placed.

[image]

G-Pereira, 2021-07-01:
Done thanks!

closed 2021-07-01 by G-Pereira