opulo-inc/lumenpnp #294: Missing BOM mobo_rev001

maximius, 2021-04-28:
First of all, I want to thank you for the great work done! This is very cool!
I want to repeat your project, including the assembly of the mobo rev001 motherboard. I was looking for information on the rev001 board(the only difference between rev002 and rev001, as I understood, is the vacuum sensor was added). I want to repeat rev001, but I did not find a BOM for this board(gerber found it on PCBWAY). Will there be a BOM for this version of the board? Will it be possible to connect feeders to work with this board?

ocouch, 2021-04-28:
Rev3 is nearly finished, I'd just wait for that. I believe the BOM was taken down because it was very out of date and no longer correct.

maximius, 2021-04-28:
Where can I find information about the new version? Only on github?

G-Pereira, 2021-04-29:
Join us on discord! The link is in the README (main page)

closed 2021-05-04 by G-Pereira