opulo-inc/lumenpnp #295: [MOBO] - Rev 3 D3 Footprint or part number incorrect?

Phase4logic, 2021-04-28:
I was looking up the footprint for the Shottky diode D3 Part Number SS34 on the Mouser web site [datasheet](https://www.mouser.com/datasheet/2/427/ss32-1768383.pdf) and it shows a SMC (DO-214AB) footprint for the part.  The footprint in the layout is a D_SMA which is a LOT smaller.  

Is the part number in the schematic correct?  The SS34 is so big it won't fit on the board.

daveismith, 2021-04-28:
> Assigned #295 <https://github.com/index-machines/index/issues/295> to
> @daveismith <https://github.com/daveismith>.

The specified part from LCSC is a DO-214AC package. This seems to also be
known as an SMA part.

daveismith, 2021-04-28:
Digikey at least carries them in the SMA package size.

G-Pereira, 2021-05-04:
SS34 is a diode series and not a complete PN. We are not yet defining a Mouser PN.

G-Pereira, 2021-07-01:
Fixed!

closed 2021-07-01 by G-Pereira