opulo-inc/lumenpnp #301: FDM vs FFF

darigovresearch, 2021-05-01:
A note that FDM is a trademarked term versus FFF for Fused filament fabrication

If you wish to change this you would need to update the [Docs page](https://docs.indexmachines.io/docs/reference/fdm/) and all the filenames in the repository itself

See the Wikipedia page for reference - https://en.wikipedia.org/wiki/Fused_filament_fabrication#Fused_deposition_modeling

ocouch, 2021-05-02:
Even formlabs uses FDM: https://formlabs.com/asia/blog/fdm-vs-sla-compare-types-of-3d-printers/

Stratasys don't state the jurisdiction of their trademark, and if push came to shove, I think they'd have a hard time arguing it's not a generic term.

https://en.wikipedia.org/wiki/Generic_trademark

Stratasys are making no attempts to enforce it - and they're probably well aware at this point it'd just be a bad look and not worth it because of how generic a term it is. Tbh, if it were me, I'd worry about it if it's a problem.

G-Pereira, 2021-05-04:
Since the FDM term is widely used using FFF would be confusing but thanks for pointing it out!

closed 2021-05-04 by G-Pereira