opulo-inc/lumenpnp #307: [FEATURE REQUEST] Add documentation preview on pull request

darigovresearch, 2021-05-08:
Some open source repositories have a cool feature that when you make a pull request it automatically generates the documentation page as a check in the pull request so people can verify the updates to the documentation work as required. Is this possible to do?

Makes it much easier to check without needing to install & build the docs manually on the user's & maintainer's laptop.

Example of good repositories that do this - https://github.com/ONEARMY/academy/pull/120

darigovresearch, 2021-06-20:
For additional context they use netlify but there are other options mentioned however not sure if there is one for docsy

https://github.com/ONEARMY/project-kamp-academy/issues/10#issuecomment-864590126

sphawes, 2022-05-11:
Regenerating the docs that frequently can be costly in Github CI bills. For now, we'll have it such that folks that are making edits should check it locally on their machine, or for small things that do not require a rebuild of the docs (typos, small simple edits), no regeneration is necessary. Any errors will also be corrected before the doc branch is merged into master.

closed 2022-05-11 by sphawes

darigovresearch, 2022-05-14:
Sounds good!