opulo-inc/lumenpnp #314: [CAD] - Suggestion - split front legs in half

stuartpittaway, 2021-05-16:
Given that the front legs take several hours to print, I suggest that consideration is given to split the leg in half (horizontally) similar to the images.

I see this having benefits such as:

1. Parts can be printed on smaller printers
2. Parts can be printed simultaneously if multiple printers are available
**3. As the top half of the part is likely to change over time, several hours can be saved by not reprinting the bottom "feet"**
4. The structure retains its strength via the 20x20 rod, with additional bolts being used for the foot and the top section

[image]

In this image, an M5 bolt could actually run vertically down the part to secure onto the top of the 20x20 rail.  Alternatively if the part can be widened, it could bolt onto both left and right sides of the rail (this may reduce width for part feeders).

[image]

maghostmil, 2021-05-23:
It looks good. Will you include Freecad or STL files for that? I can test this solution.

stuartpittaway, 2021-05-24:
I'm happy to provide the cad files, if approved

sphawes, 2021-10-05:
I'd be worried about losing rigidity that we retain by keeping everything in one print. Also, this is quite an old ticket and lots has changed since and quite a bit has been tested to be working using just one part for the leg, rather not throw a curveball into the mix now. I'm going to close out!

closed 2021-10-05 by sphawes