opulo-inc/lumenpnp #320: [CAD] - 3d printer calibration for index prints

stuartpittaway, 2021-05-20:
It would be useful for end users to be able to print a small shape containing the frequently used holes/parts of the Index machine to ensure their 3d printer is configured correctly before embarking on printing 48hours+ of real parts.

This way they can check M3/M5 tolerances and 20x20 extrusion holes

[image]

sphawes, 2021-10-05:
@stuartpittaway I saw that you closed your PR around this change, @SyberxSpace can you upload a version of that killer calibration file you designed? would love to print it out and give it a whirl!

lucian151, 2022-11-14:
Closed with the creation of the corndog model developed by @SyberxSpace a while ago.

closed 2022-11-14 by lucian151