opulo-inc/lumenpnp #326: [DOCS] Update assembly branch from master

stuartpittaway, 2021-06-17:
master branch has newer STL parts, need to sync assembly branch and documentation to update FDM images/tables

G-Pereira, 2021-09-16:
We made this manually the first time to speed things up because there was a lot of people building it and we were answering the same questions over an over again. During that tedious process we've swore the next time will be fully automatic so it could evolve over time automatically. Now to do that we've decided to move to A4 in order to use the the render workbench and place camera views on the design and make a script that used those views.to take images.
That.whole process is very time consuming so no one has started it at this point because we are working on other things.

stuartpittaway, 2021-09-16:
Ah, I didn't mean those, there is a section called FDM reference which has screenshots of each part.

This is automatically generated by a script, part of a GitHub action, but I think it needs running manually.

goncalo-pereira-builtcolab, 2021-09-17:
Why?

G-Pereira, 2021-09-17:
Why?

daveismith, 2021-09-17:
It needs to be run manually because the CI setup is a mess and we need to properly get all the pieces sequenced and running together with the right triggers now that all the various CI items have made it into the same branch.

daveismith, 2022-02-01:
I merged this into the publish site workflow so I think it's addressed. @stuartpittaway, can you take a look and close this if you think it's handled by my two PRs relating to releases.

sphawes, 2023-05-15:
closing this as we've moved docs to a separate repository.

closed 2023-05-15 by sphawes