opulo-inc/lumenpnp #327: [CAD] Two mistakes (?)

arturo182, 2021-06-18:
I think I found two small issues in the assembly:

1. The screws used for the NEMA11 are M3:
[image]

but should be M2.5:

[image]

2. There are two M3 screws on top of each other for the camera up mount:
[image]

and

[image]

This also affects the BOM which specifies 5pcs of M3x30, but in reality only 4 are needed.

lucian151, 2021-07-26:
We should switch the NEMA 11 mounting screws to ISO4762 Screw M2.5x8 per this feedback. Good catch!

sphawes, 2021-09-20:
This has been fixed, closing!

closed 2021-09-20 by sphawes