opulo-inc/lumenpnp #333: [CAD] - bolt/machine screw lengths

stuartpittaway, 2021-06-29:
Most of the screw lengths are a perfect fit when viewed on the Freecad model, however in real life, the bolt only just reaches the end of the nut.

If you use nylon/nyloc nuts, then there isn't enough thread to reach the end of the nut.

I suggest an extra 5mm is added to most of the 40mm M5 screws

G-Pereira, 2021-07-03:
Screws are supposed to just reach the end of the nut, I am in the process of printing the new parts and will validate this

stuartpittaway, 2021-07-03:
This is particularly relevant if the use of NYLOC style nuts is used - these are typically longer than standard nuts, and the thread won't reach the end of those.

lucian151, 2021-09-08:
I'm trying to do a BOM freeze for the Index Fastener kits, we might want to revisit this ticket before then.

closed 2022-03-09 by stuartpittaway