opulo-inc/lumenpnp #334: [CAD] - FDM_0011 / 0012 - poor access to tension screws

stuartpittaway, 2021-06-29:
Whilst the design of FDM11 is clever and allows the idler pully to be fitted, once it is assembled, you can no longer adjust the tension on the gantry without dismantling most of the Z parts.

In addition, the captured nut on part FDM12 has the same problem where the camera obscures access.

I have found that it is difficult to get the tension correct first time, so after a period of run in the machine gets loose and needs tightening, this is difficult as most of the Z parts/camera have to be removed.

As drawn, the slot for the V wheel is almost at its full extent of travel, which limits the possible adjustment.

sphawes, 2021-09-20:
@stuartpittaway I totally agree. A thought I've been having about the X gantry is increasing the roller count to four. This does a couple things:
- adds a bit more stability to the X gantry, preventing rotation about the x axis
- if the two bottom rollers are placed directly beneath the top rollers, there's plenty of space in the middle of the part to mount the z idler without collision

if this is something you'd like to tackle, go for it! if not, I'll take a crack at it.

stuartpittaway, 2021-09-20:
Not so sure about the 4 rollers to be honest, seems like a lot of 3d printers have moved away from that design. 

How about turning the whole thing upside down? One wheel at the top? So you tension the top when instead?

I seem to get twist of the X carriage in the Y plane unless the wheels are really tight.

stuartpittaway, 2021-09-20:
If you went with 4 wheels could the bottom 2 pivot in the middle, otherwise trying to get tension equal across both is likely to lead to uneven loading and squiffy angles.

I'm not a mechanical engineer though!

stuartpittaway, 2021-09-22:
I've had a play with adding a 4th wheel.

I've added a new piece between the front and back gantry pieces - sort of "H" shaped on its side.

I've not printed out the parts yet, and lots more tinkering to do.

Let me know your thoughts.

[image]

[image]

closed 2022-03-09 by stuartpittaway