opulo-inc/lumenpnp #335: [BOM] - mobo-rev3 branch Ring Light BOM error (or wrong footprint)

arturo182, 2021-06-30:
In the `mobo-rev3` branch the Ring Light BOM specifies the LCSC parts for the connector to be C146126 and C141812, which are 2.54mm and 2.00mm pitch respectively, however, the connector footprint is 2.00mm pitch.

Reference: https://github.com/index-machines/index/blob/mobo-rev03/pnp/pcb/ringLight/export/ringLight-bom.csv

The footprint specifies this connector: https://lcsc.com/product-detail/Wire-To-Board-Wire-To-Wire-Connector_JST-Sales-America-B3B-PH-SM4-TBT-LF-SN_C265082.html which, upon closer inspection, can't be the correct one because of the way the ring light mounts into the assembly:

[image]

It has to be a side entry connector, not top entry. If I'm correct, that would mean it should be this connector: https://lcsc.com/product-detail/Wire-To-Board-Wire-To-Wire-Connector_JST-Sales-America-S3B-PH-SM4-TB-LF-SN_C265101.html

On the other (third?) hand, is the pitch of the footprint correct? I wonder because on the mobo side the connector is a 2.50mm pitch XH-series.

I understand that it's a WIP branch so those values might be in there as a placeholder "like this, but not exactly", but I thought it might be good to have an Issue for this anyway, just so it's fixed eventually.

G-Pereira, 2021-07-01:
Thank you for pointing this out! I have both these connectors on order and I will check if they fit both the PCB and the print and decide what we'll do :)

arturo182, 2021-07-01:
Sorry, I wrote the pitch for C141812 wrong, it's 2.50mm, so neither of the two will fit the 2.00mm pitch, but if you have them in post then yes, good to validate 100%

stuartpittaway, 2021-07-02:
My LCSC delivery has already arrived with the parts in it - but looks like I didn't order the connectors for the ring light!

As a work around, it should be easy to solder wires directly to the ring light board.

G-Pereira, 2021-07-03:
@stuartpittaway I would say that is possible but you are risking tearing the pad from the PCB

daveismith, 2021-07-03:
You may want to use some hot glue and a zip tie to provide some strain
relief if you do that.

arturo182, 2021-07-12:
I have now received the side entry connector https://lcsc.com/product-detail/Wire-To-Board-Wire-To-Wire-Connector_JST-Sales-America-S3B-PH-SM4-TB-LF-SN_C265101.html and can confirm it does work with the case but doesn't match the footprint on the PCB. I mean, you can force it if you want it, but it's clearly a top-entry footprint:

[image]
[image]
[image]

G-Pereira, 2021-07-28:
Confirmed this is not an issue.

arturo182, 2021-07-29:
Can you expand, cause the footprint doesn't match the part specified 😅

G-Pereira, 2021-07-30:
Right, although it is not an issue I will keep this open to denote the change

G-Pereira, 2021-09-10:
This has been fixed by changing the connector to a XH through hole!

closed 2021-09-10 by G-Pereira