opulo-inc/lumenpnp #336: [MOBO] - Move to on-board stepper drivers (TMC2209)

G-Pereira, 2021-07-01:
**Motherboard Revision:** REV04

**Enhancement Description:**
- Move to on board drivers being TMC2209 or better at the date of this enhancement.

stuartpittaway, 2021-07-07:
Is this change to put soldered on TMC drivers on the mobo ?

daveismith, 2021-07-08:
I think that's what's being proposed. It won't be happening until mobo rev3 is released (ie, even if there needs to be corrections to rev 3 it won't include moving to onboard drivers).

I'm unsure what to think of moving this way, but would note that it will need some significant work on the heat dissipation and EMI aspects of the layout.

stuartpittaway, 2021-07-08:
Having burnt out a few driver chips in the past, a pluggable design works best in my mind.

Additionally when new chips are released, they could be used without changes

plasticuser, 2021-07-08:
I avoidf buying boards where the controller is built in. They're a pig to repair quickly in a production environment, where as driver boards of the standard fit are a quick swap. This makes fault diagnosis really quick and easy.

I'm also not keen on the idea of being locked into a specific driver. Mid-scale has a large "at home" component, and the newer generation of silent drivers are really attractive to people like me.

My $0.02.

G-Pereira, 2021-07-14:
This change is aimed at the moment we can look at the index electronics as a black box. You generally don't swap drivers on your washing machine or any other home appliances just like you won't on a final product. 

For this to be true a careful design has to be made like @daveismith suggested and some headrooms and protections need to be in place. This won't make sense during prototyping for sure.

I didn't always had this opinion but other people helped me realize that.

daveismith, 2021-07-14:
It would be interesting to do a cost calculation to see if there is much savings in BOM cost by going with the on board drivers. I would tend to agree with the sentiments of @stuartpittaway and @plasticuser that from a repairability perspective, the modules are better. If they aren't significantly more in terms of cost, it's probably worthwhile keeping modules. If the concern is soldering the headers for the modules, then we could look at SMT headers so they can be reflowed

G-Pereira, 2021-07-14:
My experience has been a little different. My 3d printer always was a self made machine and the one of the reasons I've burnt stepper drivers was when I was setting their current limit really high so I could cover up unrelated mechanical issues on my machine when I lacked experience on the subject overheating the drivers considetably. The other reason was connecting the stepper drivers reversed by mistake.

None of this reasons were a lifespan issue about the drivers and I am pretty confident that most failures people have with drivers is using them out of spec, overheating them.
Trinamic drivers aren't low quality drivers so they don't simply fail since they have several protection mechanisms. I am aware of people burning Trinamic drivers is common on TMC2130 since they have a high RDsOn which makes people overheating them most of the times.

In the Index context, contrary to "universal" 3d printer boards, we have a clear notion of what hardware we will interact with this we can design accordingly. And keep in mind that this won't impose limitations to custom setups because TMC2209 has an high current specification (2A).

EngelenH-kpn, 2021-07-29:
Which begs the question, and that is all it is, should you go for a black box design?

For the steppers, I see no benefit at all in making them onboard. Maybe I am missing it. Even if you design the board for just the Index hardware, accidents happen. Also, given the target audience for this device, I think I can say with some certainty that this audience will have a bin full of the plug-in stepper drivers on their desk right now.

On a larger whole, and this is a question more for me, would it not be an idea to build the board as a 'universal' board? Or at least to some extent. Yes it would be mainly for the 'index' hardware but as you already indicated. People will quickly start making custom setups, might as well embrace them into the community too. For instance I recently scored a TVM802 cheap, could see myself using this board for that machine too.

bholmes451, 2021-07-29:
I believe there are plenty of "Universal" boards on the market already (I have several). Some have stayed the course and are still around and other have not. I agree with @G-Pereira on this. There is not much advantage in going universal for this simply because we know the hardware targeted (Pick and Place machines). If you are looking at using something outside the design, then perhaps a different board would be better. At most, I would agree to split out the Step/Dir pins for each stepper driver so you could send them to a separate, not on board, system. I do this sometimes, to add industrial servo motors to my design.

daveismith, 2021-07-29:
> @EngelenH-kpn <https://github.com/EngelenH-kpn> moving to on board
> drivers drastically decreases assembly costs by helping to reduce THT
> components. One of REV04 goals is to make it completely SMT, which makes
> even more sense on a PnP motherboard. You have a PnP but the motherboard
> isn't capable of being completely assembled by it? Weird.
>
> In my view, a way to mitigate accidents is to reduce user input. For the
> motors we are targeting (nema17) this TMC2209 drivers cover it's operating
> range very well (it even has some headroom) so there's no reason to worry
> about upgrading drivers which is, in many cases, just a form of consumerism.
>
> That being said, imagine if you would actually need to use a different,
> more beefy, kind of motors, there's other components and board
> characteristics (layout) that will have bottlenecks. There will always be
> bottlenecks, nothing is really universal in that sense.

Didn’t we land on the approach to this ticket being a review to see if it
would be possible to handle both and a user can choose which to populate.

In terms of headers, you can absolute get SMT headers which the drivers
could use so I don’t think the argument of putting down the driver itself
being the only way to get to fully SMT applies (also we have USB ports and
the power terminal block which shouldn’t go SMT for mechanical reasons).

Having said that, putting the driver on board would reduce component count
and likely make the overall solution cheaper.

G-Pereira, 2021-07-30:
This is something we have to discuss yes. I am just answering "what advantages would it have"

EngelenH-kpn, 2021-07-30:
I guess from my end, but it is not my project so if this is the case so be it, I am surprised or perhaps more accurately disappointed the mission goal for this project (or at least the controller board) has changed. That was one of the reasons why I took notice of the project over a year ago.

If the costs are not much reduced as you say, we are left with just the complete assembly by JLCPCB. Which also touches upon one other reason for me personally to be weary of this black box approach you spoke of. I have the displeasure of living in a region where availability the components (mechanical and electronic) are just a pain and import fees are oppressive (especially for China products, yes that fun EU law that went into effect on July 1). So for me it will have to be a case of 'use what I can find easily and often locally'.

plasticuser, 2021-07-30:
The mission statement of the Index placer is to have a machine that is affordable for the common person. Affordability isn't just the initial price. It's also the build quality, the future maintainability and overall reliability of the board, and the death of the board when it is no longer serviceable or has been superseded by a better model.

While I'd *prefer* plug in steppers so I can use the board with a wide variety of systems and have it available and useful to most of the OpenPnP community, I also will buy it anyway if only to have an official unit to test against.

The thing about machines designed by committee is we don't all get everything we want. So if the drivers are built in, I will still get one. I'd just prefer they're not, for a multitude of reasons.

EngelenH-kpn, 2021-08-02:
I agree except the 'universal' aspect, for clarity, was one of the original goals of Stephen. In fact his video as to that effect comes up (at least for me, not sure if this is just for me or not) at the top of the list when I google 'universal openpnp controller'.

So I am not questioning the change discussed here because I want to change the direction of the project and steer it towards 'universal' either personally or 'by committee' but because this change seems to conflict with what Stephen set out to do. On top of that I think he had the right idea.

Possibly however this goal has changed and I am just not aware of it.

G-Pereira, 2021-08-08:
@EngelenH-kpn that's correct, we need to discuss this more through because of that. At the moment we are excited to make the machine do what it was designed to and it is starting to show real and awesome results so this is definitely at the bottom of our priorities. I think Rev03 has been showing great results and it will be here for a while!

feldon30, 2021-08-28:
Stepper driver ICs are known for burning out. The target audience of Index PnP is not pure Consumers but Makers who would be more than comfortable replacing a module but may hesitate at floating and replacing a QFN-28 on a densely populated 4-layer board. I believe Modules would give us a bump in Repairability.

A4988 modules seem to be about $2 each.
TMC2209 modules seem to run $4.50-6 each.
TMC2209 bare chip are currently unavailable DK and Mouser, while Newark has some for $3 each. With the addition of trim pots and other support hardware, the price difference at this scale may be a wash when comparing integrated vs. module.

plasticuser, 2021-08-28:
Using modules allows us to take advantage of future stepper driver models
instead of locking us in to a specific part.

It makes it trivially repairable.

It makes it reconfigurable to future needs, experimenting and future Index
development.

Using SMD mounted components deprives us of all these

MrPix

lucian151, 2021-09-08:
Should we label this "wont fix"? Seems like the plan after a lot of discussion here.

EngelenH-kpn, 2021-09-09:
Alternatively you could support both options, put the footprints for both on the board so you can either use an integrated stepper driver or solder in headers for a module. Maybe even add something like screw terminals or a connector to hook into an external stepper driver module. It would satisfy all groups but cost space on the board.

G-Pereira, 2021-09-12:
@lucian151 I think this could be revisited once we line our ideas for rev04

sphawes, 2021-09-20:
Going to close this. I agree something definitely to be revisited, but if it becomes an obvious improvement down the road, we'll make a new issue for it then! :D

closed 2021-09-20 by sphawes