opulo-inc/lumenpnp #344: [MOBO] - Buck Regulator/Dropout Regulator/Power Path

EdHayes3, 2021-07-07:
**Motherboard Revision:** 
REV003

**Issue Description:** 
- Some placement and traces around buck regulator should be adjusted to match guidance in datasheet for PCB layout.
- Not enough thermal relief for the regulator
- Traces are quite thin for some of the power path/rails
- Only two small spokes for the entire 12v rail (motors, mos1-4, aux 1-3.)
- Inductor has a low current rating at 2.2a (assuming the goal is 3a output since that was the output of the buck).
- The bottom ground plane is almost entirely cut off from the input for most of the board. Seems one of the internal planes is ground too... Internal layers are thinner than the outer layers. Maybe someone with more experience here can chime in if this is even a big deal.
- The dropout regulator for 3.3v is connected to the thermal vias by way of spokes, that defeats the purpose there.

**Suggested Fix:** 
- Much of this is in the datasheet for PCB layout, but here are some of key points.
- Power path should be direct as possible around the regulator 
- Ground connections for input, output, Buck, and diode should be direct and top side of PCB.
- Ceramic input capacitor should be closer and direct to regulator, not via a via. 
- Add a larger value ceramic capacitor too if possible. Buck regulators are much more efficient with ceramic capacitors.
- Ground pin is where heat is dissipated from regulator, add more thermal vias to bottom ground plane.
- Might also want some vias around the catch diode. I see four to the one side of it.. not very close to it though.
- Inductor should be 20% higher current rating than expected current requirements.
- Inductor should be closer to the buck regulator and ideally not have the trace go under the regulator.
- Thicker traces for power path to reduce impedance. Eliminate the spokes and do direct connections for some of the major parts like the bulk power capacitors, p-mosfet, catch diode, and connections the power sources.
[image]

**Other comments**
There's other footprints of regulators that seem to be more common to reduce reliance on a single part with limited stock (Footprint 1: C115136, C158090, C170097, C329268; or Footprint 2: C9388, C32078, C71111, C126079 --These support 24v. There's other common footprints that have even more interchangeable parts if 12v is the input expectation, I can expand on this if requested.) Picking a synchronous regulator which eliminates the catch diode could save some space, BOM count, and increase efficiency.

goncalo-pereira-builtcolab, 2021-07-14:
Thank you @EdHayes3 ! These are all good recommendations which we are looking to target to mobo rev04. An important part of a PCB design is to minimize, like you have mentioned, the unique part count. Since rev03 aimed at improving the electrical design of rev02 and include some new features it is not yet worthwhile to do layout and MPN optimizations in this pass provided the design is already in a very good state and we are still exploring some improvements in the electrical design specially for vacuum sensing. Rev04 will target precisely that as well as reducing assembly costs.

G-Pereira, 2021-07-14:
Thank you @EdHayes3 ! These are all good recommendations which we are looking to target to mobo rev04. An important part of a PCB design is to minimize, like you have mentioned, the unique part count. Since rev03 aimed at improving the electrical design of rev02 and include some new features it is not yet worthwhile to do layout and MPN optimizations in this pass provided the design is already in a very good state and we are still exploring some improvements in the electrical design specially for vacuum sensing. Rev04 will target precisely that as well as reducing assembly costs.

G-Pereira, 2023-08-30:
The layout is being completely redone considering manufacturer recomendations