opulo-inc/lumenpnp #345: [MOBO] - Polarity Protection P-mosfet current limit/Thermals

EdHayes3, 2021-07-07:
**Motherboard Revision:** 
REV003

**Issue Description:** 
- Polarity protection P-mosfet is pretty close to it's current limit. If each stepper motor is at 1.7a, you are at 6.8amps. P mosfet is rated at 8.8A, and assumes you are using a 1 in^2 pad, with 2oz copper.  With all the other stuff on the board, I'm thinking you may want some additional current headroom and/or thermal management here.
- Obviously, rarely everything will be consuming power at all times, I'd still try and go a bit higher if you can. 
- Pulsed current is higher, but assumes duty cycle ≤ 2%; not sure if this applies.

**Suggested Fix:** 
- Alternate part (IRF9321TRPbF*, AO4407A, HY12P03S*, NCE30P25S*) [I did not spend too much time looking for alternatives]
- Watch gate-source voltage on any replacement if using 24v power supply.

G-Pereira, 2021-07-14:
We have designed to configure motors at 1A so there's more or less 1.5A also considering the remaining power consumption. I agree that's hard to guarantee without having a specific type of drivers allowed with a specific provided configuration file that limits current. We don't want to have this for now so people can prototype the machine easily at the moment with any driver they have at hand.
This will change in rev04 and a more formal design can be established.

theacodes, 2023-05-01:
This has been resolved in rev4, as far as I can tell.

closed 2023-05-01 by theacodes