opulo-inc/lumenpnp #349: [MOBO] - Reverse polarity protection mosfet is reversed

G-Pereira, 2021-07-10:
**Motherboard Revision:** REV03

**Issue Description:**
- FDS4435BZ mosfet is drain and source are reversed.

**Suggested Fix:**
- Rotate mosfet without moving footprint to keep the same stencil

Jnesselr, 2021-07-11:
I don't understand how simply rotating it would work. It looks like it is pin 4 for the gate, 1 for the drain, and 8 for the source on the schematic. On the chip itself, pin 1-3 are source, 4 is gate, and 5-8 are drain.

Rotating it 180 degrees seems like it would connect the source and gate together.

G-Pereira, 2021-07-13:
@Jnesselr I think you got it by our yesterday's conversation but I'll post here so other people can understand too.
I don't mean we can rotate the IC because like you've seen there's no orientation that would fix it, a new board spin up is needed.

G-Pereira, 2021-07-13:
I was looking at it wrong. This is correctly connected as it is on the current beta version!

closed 2021-07-13 by G-Pereira