opulo-inc/lumenpnp #359: Marlin changes for nozzles letters in OpenPnP

mariusmotea, 2021-08-05:
Hi,

I build a PNP machine based on this project but with some changes from my side on electronics part. I used SKR 1.4 Turbo with TMC drivers so i cannot use the code from here. I follow and old Marlin commit for OpenPnP and manage to have the machine controls by specifying E0 and E1 as letters for nozzles, but they work only when rotation is positive. Most components on my project require rotation to a negative value and in this case the nozzle stop at 0 value and place the part exactly how it was picked from the tape. In an example from here i saw the use of the letter R for the nozzle. I test that, but it don't work at all. Can somebody help me with the changes to configuration.h and machine.xml to have the rotation working properly +/- 180 degree.

Thanks.

G-Pereira, 2021-08-08:
Hey @mariusmotea !
I don't believe using a SKR is related to the issue you're having. I have been using the new axis naming successfully with what's in our marlin fork at the index-machines GitHub organization and the OpenPnP from my PR #358 have you tried that already?

G-Pereira, 2021-08-08:
Be sure to use this branch
https://github.com/index-machines/Marlin/tree/index_marlin

mariusmotea, 2021-08-08:
I fix it by replacing extruders with independent axes. Now i use GCODEAsyncDriver in OpenPNP. The code from there use only 3 axes so i think it can work only with basic GCodeDriver. My machine is almost ready, will post all my changes.

closed 2021-08-08 by mariusmotea

mariusmotea, 2021-08-08:
> Be sure to use this branch
> https://github.com/index-machines/Marlin/tree/index_marlin

https://github.com/index-machines/Marlin/blob/index_marlin/Marlin/Configuration.h#L166

I think that line must be uncommented and set to 4 or 5 (by case) axes.

G-Pereira, 2021-08-10:
Ah! Sorry for not saying that before but actually to be able to merge that branch into marlin we need to keep configuration.h untouched. That branch adds the board file but then you need to copy the configuration file from here:

https://github.com/index-machines/Marlin-Configurations/tree/index_config/config/examples/Index

And off you go!