opulo-inc/lumenpnp #363: [CAD] - NEMA8 fitting

jlshelby, 2021-08-24:
The current specification seems to be fit for the NEMA11 CP40 style nozzles.  The NEMA8 Juki is also a common and easy to get nozzle and would like to see an additional part with the sizing for the NEMA8 servos if possible.

G-Pereira, 2021-09-14:
This machine uses NEMA11 for several reasons and we have no plans to support NEMA8, not to say it isn't possible for people to use NEMA8 but definitely something to be available on [our Thingiverse page](https://www.thingiverse.com/groups/index-mods) once someone designs it! :smiley:

closed 2021-09-14 by G-Pereira

mbike2000ru, 2021-09-15:
> The current specification seems to be fit for the NEMA11 CP40 style nozzles. The NEMA8 Juki is also a common and easy to get nozzle and would like to see an additional part with the sizing for the NEMA8 servos if possible.

I use a sort of a converter plate, it has holes for nema 8 as well as for index. It was easier - to fix this way (no need to rebuild the whole part)

jlshelby, 2021-09-15:
I decided to scrap this whole design as it wasn't working for my needs. I've redesigned all the X/Y parts and presently working on the Z axis.  I've reduced the cost quite a bit with the enhanced design so no worries as I ended up getting a NEMA11 with the CP40s just to try out the accuracy.

G-Pereira, 2021-09-15:
We are interested in hearing about your improvements, can you share it on [our Discord](https://discord.com/invite/TCwy6De)?