opulo-inc/lumenpnp #364: [MOBO] - REV003 BOM Missing PN for Motor Driver Sockets

Phase4logic, 2021-08-24:
REV003 - Motor Driver Sockets missing from BOM

I'm not seeing an entry in the BOM for the Motor Driver sockets (A1-A6).  I can see that they are 8 pin sockets with 2.54cm spacing, but I don't know how tall they need to be to clear the capacitor and any pins sticking down from the driver board.  Could someone respond to me with a part # or specs?  

Thanks!
Mark

feldon30, 2021-08-31:
There's actually an entry in the BOM for these that says Pololu_Breakout-16_15.2x20.3mm which is a standardized stepper driver footprint used in CNC, 3D printers, and other applications that use stepper driver modules. However the note field in the BOM next to the Pololu Breakout suggests A4988 stepper drivers which are the most basic steppers you can get ($1-2/ea). According to several comments in Discord, the preferred stepper driver for the Index is the much upgraded TMC2209 ($4.50-6/ea). If I were you I'd wait for the next revision before ordering any parts.

Phase4logic, 2021-08-31:
Actually I was looking for a part number for the sockets that are soldered
onto the board that the driver you are referring to plugs into.

feldon30, 2021-08-31:
It's 2.54mm straight socket female pin header.

Phase4logic, 2021-08-31:
Do you happen to have a minimum vertical height so that the driver board clears the capacitor under it?  These sockets should be in the BOM. 

Mark

daveismith, 2021-08-31:
You can use LCSC part [C429957](https://lcsc.com/product-detail/Pin-Header-Female-Header_JILN-22850108ANG1SYA01_C429957.html). I wasn't sure how to put it into the BOM because it's a single footprint but you need to order two of this part for each footprint.

Phase4logic, 2021-08-31:
Thanks David. One approach would be to make the socket the BOM item, which would split the footprint into two parts. (I’m not sure how to do that) Then the next-level-up BOM would have the PCB, the motor drivers and any other items that plug into the board.

G-Pereira, 2021-09-02:
I was aware of this but I still haven't found a clean way to assign that part number to the PCB. I am avoiding using two separate footprints for the driver headers since that is easily messed up while working on the layout.

Let's keep this issue open until this is sorted out or maybe receive suggestions from anyone.

solarmax1, 2022-03-20:
I happened upon an article that discusses how to handle this in KiCad. The examples they give are the opposite of this case, but I think you will get the idea.   Wow, I just noticed this article is from 10 years ago!  Hope it still works.
https://kicad-users.yahoogroups.narkive.com/NXR3fWfE/managing-multi-part-components-ic-socket-battery-socket-into-bom

theacodes, 2022-06-19:
KiCAD 6 allows items on the schematic (and therefore BOM) to be excluded from the PCB. We can use this to note the MPN for the sockets.

PR #664 (2023-05-12): Update mobo schematic with part numbers and ratings
All parts now have part numbers for at least one reseller or manufacturer along with rating specifications.

Note: I did not update the PCB with these new properties, as it seems the PCB is a bit out of sync with the schematic.

Closes #646
Closes #626
Closes #364
Closes #315
closed 2023-05-12 by sphawes