opulo-inc/lumenpnp #375: [MOBO] MOBO BOM File "mobo_bom_jlc.csv" column mislabeled

Phase4logic, 2021-09-06:
L1 in the schematic has the LCSC Part Number field populated with C168077.  However, the LCSC part number field in the BOM file [here](https://github.com/index-machines/index/blob/master/pnp/pcb/mobo/export/mobo_bom_jlc.csv) is blank. 

I think that the column in the BOM file is mislabeled.  It is labeled "LCSC Part #" and I think it should be labeled "JLC PCB Part #".  I see sometimes they are the same part # and sometimes they are different.  In this case, the inductor has no JLC PCB Part # filled in, which would explain why it is blank in the BOM.

daveismith, 2021-09-06:
JLC required that column to be LCSC when the export scripts were developed. It seems like they now suggest JLCPCB Part #. 

See https://support.jlcpcb.com/article/80-bill-of-materialsbom-file-for-smt-assembly

martinbogo, 2021-09-06:
They aren't one and the same?

daveismith, 2021-09-06:
We have separate attributes in the schematic as not all LCSC parts are available at JLCPCB. As such sometimes there's a cheaper option on LCSC that we can't use with JLC.

G-Pereira, 2022-01-24:
This issue has been solved since those files were removed and the existing BOM was updated (now via an automated script)

closed 2022-01-24 by G-Pereira