opulo-inc/lumenpnp #397: [MOBO] - Cost vs Effectiveness Issues

ArcAIN6, 2021-10-03:
**Motherboard Revision:** What motherboard version is this enhancement suggested for. If you have one, this should be on the front of your board in the format REV### or REV###-???????

**Enhancement Description:** Talk about the enhancement you want.
- Does it enable a new feature? Yes
- Will it affect existing features? No
- Have you prototyped it successfully? Haven't attempted
- Attach relevant schematic / pcb drawings.

As this is your project, i feel it's best if you do any alterations to it, 

However, the main mobo's are excessively large needlessly. 
There is a LOT of empty space on those boards, and the cost of ordering 2 of them, per feeder is unjustifiable. 
Through PCBWay, the boards would cost me $67 for 5 of them, and you need 2 per feeder, that's getting a bit expensive if you're talking 10+ feeders. 

Add to that the unusual shape of the PCB's, and you actually paying for cut-away space. 

I feel it would be more economical, and better for the project as a whole, if these were split into smaller boards, and connections made between them using either FFC, or ribbon cable. 

Considering the fact that the majority of the feeder is 3d printed anyway, it makes little to no sense to use expensive circuit boards from a fab as a structural component, when 3d printer filament is much cheaper by volume. 
If you're worried about the longevity of the feeder over time of placing, and replacing them, a simple metal sleeve, or strike plate, attached to the printed part in the area of contact with the rail would be more than sufficient, and combined would still be more cost effective. 

An added bonus of splitting the board into multiple smaller boards, is the fact that any design changes, due to new features, or a change in part technology, would mean for a less costly order of new boards, as you wouldn't need to order all of the boards, if say just one IC on the main board were to be changed.

G-Pereira, 2021-10-03:
Hey! We agree with you and we're working on it for the next revision.

lucian151, 2023-03-25:
Closing this issue out as newer revisions of the feeder are MUCH smaller.

closed 2023-03-25 by lucian151