opulo-inc/lumenpnp #404: Standardize Print Dimensions for Common Hardware

sphawes, 2021-10-05:
Please fill out the following:
- Build Number where this issue is present. (This is the same as the most recent tag number): v1.2.2
- Is this recording an active issue, or suggesting an improvement?: Improvement
- Are you just logging an issue, or do you have a solution in mind?: Solution
- Please describe the issue:

As of now, there is no central reference for how large of a void we make for, say, a captive M5 nut. This results in slightly varying fits across the machine.

- Please describe your suggestion solution, if you have something in mind:

Add a section to the Contributing page on the wiki outlining standard dimensions for:

- M3/M5 thread-in diameter
- M3/M5 pass through diameter
- M3/M5 Socket Head screw clearance height/diameter
- M3/M5 captive nut profile and depth
- 20x20 extrusion profile press-fit
- Others I might have missed, but are used frequently and should be standardized

G-Pereira, 2022-03-03:
I commented and closed the wrong issue by mistake, reopening this

sphawes, 2023-05-15:
This has been completed! closing

closed 2023-05-15 by sphawes