opulo-inc/lumenpnp #417: CP40LV-N75 nozzle does not fit into FDM-0024-02 nozzle holder

enwi, 2021-12-04:
- Build Number where this issue is present: 
dd1de7f7706633517650f5877f8b2a5a7f15347c 
- Is this recording an active issue, or suggesting an improvement?: 
Issue 
- Are you just logging an issue, or do you have a solution in mind?: 
logging an issue
- Please describe the issue, including relevant part numbers and screenshots of the issue in CAD if possible:
If you take a look at the pictures of the nozzles you can see that the `CP40LV-N75` (the one on the far right) is a little bigger than it's base. The `FDM-0024-02` nozzle holder does not account for that. The nozzle holder space is 6mm wide, but the nozzle is 9mm wide (diameter).
[image]
[image]
- Please describe your suggestion solution, if you have something in mind:
The `FDM-0024-02` nozzle holder needs to be widened in the 6th position for `CP40LV-N75` to fit.

stuartpittaway, 2021-12-06:
@enwi are you able to grab a new copy of this from the cad-staging branch, print it and let me know if its working for you?

I've had to move the bolt mounting a few mm, so your existing OpenPNP config may need tweaks to the nozzle auto changer settings.

enwi, 2021-12-06:
> @enwi are you able to grab a new copy of this from the cad-staging branch, print it and let me know if its working for you?

I will print it later and let you know

enwi, 2021-12-06:
@stuartpittaway Fits perfectly now 👍

closed 2021-12-06 by enwi