opulo-inc/lumenpnp #450: McMaster-Carr link errors in BOM

qwertymodo, 2022-02-17:
Minor issue, but McMaster-Carr links seem to require a forward slash at the end of the URL, or else they redirect to the homepage instead of the intended product page.  There are a handful of McMaster-Carr links found in the BOM that are missing the trailing slash, e.g. https://github.com/index-machines/index/blob/master/pnp/INDEX%20BOM%20-%20MECH.csv lines 56 and 59

qwertymodo, 2022-02-17:
Hmmm... this may actually not be the case.  It just happened to me a minute ago, but now I can't reproduce, with those exact same links.  Maybe it's just a fluke on my end.

stuartpittaway, 2022-03-09:
@qwertymodo the BOM was updated significantly by the last release, can you test again and let me know?

qwertymodo, 2022-03-09:
I'm starting to think it might have just been an intermittent issue server-side, and I'm unable to reproduce the error again. So, I think at this point we can just call this a non-issue.

closed 2022-03-09 by qwertymodo