opulo-inc/lumenpnp #451: 8mm Strip Feeder

DirtyEngineer, 2022-02-18:
There has been several requests for a 8mm version of the strip feeder. The parametric file has several issues with the 8mm value. The male tabs are 8mm to long and the 3 relief holes are 4mm from were they should be. I sorted those out and exported the STL's for the strip feeder and cover which are attached. I have not printed these so please use at your own risk. 

[image]

[8mm Strip Feeder and Cover STL.zip](https://github.com/index-machines/index/files/8099309/8mm.Strip.Feeder.and.Cover.STL.zip)

Also included is the FreeCAD part. You will need to set the correct path to the "RobotoMono" font from the Lumen repo. Do not try to use this file for the other strip sizes as the formulas are not correct.

[FDM-9002_8mm.zip](https://github.com/index-machines/index/files/8099355/FDM-9002_8mm.zip)

mbike2000ru, 2022-02-19:
[image]
Hello,
I made a 8 mm smd feeder guides and a cap just to keep smd components after work. But I did it in Fusion360. 
I can give my STL if needed. 
I attach it to the mounting plate simply by squzing by 4 clamps. It allows to use other sizes of feeders.

stuartpittaway, 2022-02-21:
@G-Pereira can you with the existing parametric strip feeder?  Seems like a lot of people are having issues after the "keeper" was added.

stuartpittaway, 2022-02-21:
@DirtyEngineer just to clarify, you have taken a look at the "StripFeederParams" spreadsheet in FreeCad and tweaked the values (the ones which state "ONLY CHANGE THIS!").

You can set "TapeWidth" to 8mm and "NumberOfHolders" to the required value (8)

[image]

This should then produce this file.  You may need to tweak the font file/path on the "Cover_Label" and "SMD_Material" objects.

The tabs should have a hole in them as well, they are designed to allow multiple strip feeders to be placed next to each other without requiring gaps/taking up extra space.  They can also be disabled by changing the parameter "EnableTab" (although looks like this has some issues at the moment)

I would also recommend printing this piece flat on its base - the version in releases appears to be set vertically.

[image]

G-Pereira, 2022-02-21:
Thank you! I'll review this in the next day or two. I have a bunch of these printed and can better see what is going on.

InfiniteNesLives, 2022-02-23:
Stephen shared the one he's currently using on discord.

[image]

[FDM-0048_8mm-strip-feeder.zip](https://github.com/index-machines/index/files/8125289/FDM-0048_8mm-strip-feeder.zip)

DirtyEngineer, 2022-02-25:
Seems this isn't really an issue. The model has the clearance holes placed so they align with screw heads on the staging plate.  The tabs are zero thickness so they don't print so that issue isn't a problem.

Attached is an 8mm x 7 lane STL as this seems to be a common requested size.
[FDM-9002_8mm_x7.zip](https://github.com/index-machines/index/files/8142478/FDM-9002_8mm_x7.zip)

closed 2022-02-25 by DirtyEngineer

G-Pereira, 2022-03-06:
@MakeItHackin @InfiniteNesLives and @mbike2000ru The parametric strip feeder is meant to be flexible with every user preferences. Keep in mind that that feeder Stephen shared only allows resistors and capacitors and there's some components that use 8mm tape and the pockets are a little deep