opulo-inc/lumenpnp #458: Z Gantry Left (FDM-0017)->Linear Rail: Screw length incorrect?

jonnerd154, 2022-02-26:
Please fill out the following:
- Build Number where this issue is present. (This is the same as the most recent tag number): v2.0.1
- Is this recording an active issue, or suggesting an improvement?: Active Issue
- Are you just logging an issue, or do you have a solution in mind?: Solution in mind
- Please describe the issue: Instructions indicate to use four M3x10 screws. Threaded hole in the linear rail is too shallow to accept a 10mm screw.
- Please describe your suggestion solution, if you have something in mind: Switch to M3x8mm?

sphawes, 2022-03-01:
Hey @jonnerd154! Thanks for the note!

Just checked on my machine, I'm using all M3x10mm machine screws for mounting the Z axis. Just to check, are you threading the screws through both the Z gantry back plates, along with the Z gantries? The total stackup of those two prints should be enough to prevent the screw from bottoming out. See the image below for reference.
[image]

jonnerd154, 2022-03-01:
Hey @sphawes Happy to help! Thanks for your attention. I am passing the screw through both back plate and the gantry (and the machine is running nicely with 8mm screws in there!). 

There was some discussion on the Discord about this, seems others are having a similar issue: maybe the depth of those screw holes isn't well controlled at the factory? To be fair it's not clear that this JesseL was screwing through both components. https://discord.com/channels/688723670461448225/814870897722523699/947559958684254239

My machine is running a job right now, but when it's idle I'll take the Z-gantry off to measure the hole depth of the linear rails.

commit de3244b0 (2022-03-08): fixes #458. updates screw length for mounting z axis so some screws dont bottom out