opulo-inc/lumenpnp #459: Ring Light PCBA: Mech interference between FDM-0015-02 and C45

jonnerd154, 2022-02-26:
Please fill out the following:
- Build Number where this issue is present. (This is the same as the most recent tag number): v2.0.1
- Is this recording an active issue, or suggesting an improvement?: Active issue
- Are you just logging an issue, or do you have a solution in mind?: Solution in mind
- Please describe the issue, including relevant part numbers and screenshots of the issue in CAD if possible: When installed inside FDM-0015-02, the capacitor at the edge of the board (C45) interferes with the inner wall of FDM-0015
- Please describe your suggestion solution, if you have something in mind: Cut a small relief in FDM-0015 to provide clearance to the component.

For a quick fix on already-printed parts, some xacto-ing works. :)
[image]

TheSeven, 2022-03-01:
The same issue applies to FDM-0019 as well.

stuartpittaway, 2022-03-02:
FDM-0019 is fixed by issue #471

stuartpittaway, 2022-03-02:
Updated part now has cut out for C45 capacitor and also mouse bite edges.  Removed alignment notch as that now conflicts with mouse bite slots.

Updated to part rev. 3

[image]

sphawes, 2022-03-08:
Fix released in v2.1.0!

closed 2022-03-08 by sphawes