opulo-inc/lumenpnp #461: Mech interference between Ring Light PCB and set screw on camera lens

jonnerd154, 2022-02-26:
Please fill out the following:
- Build Number where this issue is present. (This is the same as the most recent tag number): v2.0.1
- Is this recording an active issue, or suggesting an improvement?: Active i
- Are you just logging an issue, or do you have a solution in mind?:
- Please describe the issue: On the cameras shipped with the early bird kits, there is a small setscrew on the side of the lens. Unfortunately it's _exactly_ at the elevation of the Ring Light PCB, and protudes far enough to prevent correct alignment of the camera in the down-camera module.
- Please describe your suggestion solution, if you have something in mind: 
  - Short term: adjust documentation to suggest removing that screw (this could be a bad idea)?
  - Long term: cut a notch in the board outline of the Ring Light PCB.
 
[image]

stuartpittaway, 2022-03-01:
See also #471

sphawes, 2022-03-01:
Thanks for the note @jonnerd154!

I think we haven't experienced this issue yet because I'll usually only ever back that screw out just enough so that the lens can rotate, and it hasn't been enough that I've had this collision. Some thoughts:

- Easy win for the next rev of the ring lights to have a notch here!
- In the meantime, removing the screw altogether should be alright. These lenses are in there pretty good (as I'm sure you've noticed) and it takes a bit of grip to screw them in and out. We've been running our machines consistently with that screw not engaged with the lens and haven't had any shift in focal plane.

I'll leave this open so we can make sure we add that notch on the next pass of ring lights!

jonnerd154, 2022-03-01:
Sounds good, thanks! And yes, agreed - that lens isn't going anywhere without significant encouragement. ;)

Just an additional FYI: the screw was contacting the PCB even in it's from-the-factory position.