opulo-inc/lumenpnp #463: FDM-0016 not included in Release Assets `index_bom.md`

jonnerd154, 2022-02-26:
Please fill out the following:
- Build Number where this issue is present. (This is the same as the most recent tag number): v2.0.1
- Is this recording an active issue, or suggesting an improvement?: Active Issue
- Are you just logging an issue, or do you have a solution in mind?: Logging an issue
- Please describe the issue: FDM-0016 not included in Release Assets `index_bom.md` Qty required appears to be 6.
- Please describe your suggestion solution, if you have something in mind: Something with the CI? Or maybe this is manual. I'm not sure how y'all have that setup.

daveismith, 2022-02-26:
It seems like this is getting missed because it's not in the freecad assembly file. Updating that should take care of the issue. Please note that if you're going to use `cad-staging` to do this work you need to sync it to master is it is out of date and may break documentation if you don't (it's got the old github actions configs).

stuartpittaway, 2022-03-01:
The current cad-staging now includes FDM-0016 and the new BOM build script also exports it into "index_bom.md"

sphawes, 2022-03-08:
This was fixed in the v2.1.0 release!

closed 2022-03-08 by sphawes